captain-amygdala/pistorm #10: part# for CPU connector and RPi connector

alestanga, 2021-03-12:
hello,
I've just ordered the PCBs from JLCPCB and I'm about to order the components from LCSC.
since there is no CPU connector in the BOM, do you have any suggestion about which one to buy?
I've already built an Open Amiga 500 RAM Expansion but it ended up being to tall to fit in the case so, I'd like to buy one that fits inside :)
also, for the RPi, I'm buying this one from LCSC: C350274
is it too tall or could work?
thanks.

madgrizzle, 2021-03-12:
May want to post this question on the discord channel:

https://discord.gg/qnzEXnPh

There's a lot of activity there and there was a recent discussion regarding this topic.

alestanga, 2021-03-12:
thank you, I'll join the channel.
I didn't know about there was a discord channel.

closed 2021-04-11 by captain-amygdala