captain-amygdala/pistorm #23: Make pistorm sustainable

rzr, 2021-05-13:
I wish this project to survive the greedy ones and make sure a decent retribution to the authors.

May @OSHWA , @OpenCollective or other organizations help

gianlucarenzi, 2021-05-13:
I hope @captain-amygdala  will have some contact with those nasty people to arrange something legal... Otherwise the project itself will go closed-source in the near future... WTF...

flibblesan1984, 2021-05-13:
> I hope @captain-amygdala will have some contact with those nasty people to arrange something legal... Otherwise the project itself will go closed-source in the near future... WTF...

Legally, they cannot do anything. There are no clauses in the MIT license that forbid that behaviour. Morally it's wrong, but legally, the ebay seller isn't doing anything wrong.

gianlucarenzi, 2021-05-13:
Are you referring to the UK based seller with (partially mounted)
components?

flibblesan1984, 2021-05-13:
Yes, sorry. The seller was linked on Discord as selling partially completed boards which has upset a few people.

rzr, 2021-05-13:
Thanks for feedback, I don't see any legal issue to do what ever the licence permits.
Morally this is an other question, but my point is if there is a market for pistorm devices,
then it's normal that items are made and sold
so the challenge is how this ecosystem be profitable for the community as a whole ?

gianlucarenzi, 2021-05-13:
Actually, I do not know. I made a lot of projects (all of them are
OpenSource) and I have found some people who sell my boards on ebay. (
https://github.com/gianlucarenzi ) I do not know if they charge too much or
not, it's not an issue to me.
I made my project to fullfill my needs, that's it. @claude was trying to
demonstrate to himself he is as good as the other companies' Amiga
Accelerator makers. Or simply to have a 68k cpu replacement as proof of
concept.
He stated (due to the well known TF1260 stuff) all high prices accelerator
will force to close down the project, so it's up to him.
In my opinion, I do not care if someone else is makeing money producing and
selling my stuff, as I am able to improve it by myself and put it under
another license if I would...

rzr, 2021-05-13:
ok feel free to close this ticket if it's not an issue.

Anyway while I am here,
let me share about my thoughts about FLOSS sustainability:

https://rzr.github.io/rzr-presentations/docs/abandonware

closed 2021-05-13 by beeanyew