captain-amygdala/pistorm #37: Whdload and Pistorm

ragne1, 2021-07-28:
Im getting a error when starting games with whdload -> whdload error slave is illegal becuse contains relocations, ??
Anybody else have this error?
Dennis

LIV2, 2021-07-28:
@ragne1 do you have any fast ram cards installed in your Amiga? I have seen this error when I had a Fast ram board installed while using PiStorm

ragne1, 2021-07-28:
Yes I do, Im going to remove it and se if it makes it run.
thanks for the info.

ragne1, 2021-07-28:
Ok now they start, but frezes after a couple seconds.

ragne1, 2021-07-29:
So I think that ill remove the pistorm since it really does not work.
Original installed games from floppy to hd do work but hm whdload sucks.
No whdload games run they all freze after a couple seconds.

ragne1, 2021-07-29:
Thanks, but for now I give up.

captain-amygdala, 2021-07-30:
Make sure to disable all MMU features in your global whdload config and/or tooltype.
http://whdload.de/docs/en/mmu.html
With that I run whdload without major problems on pistorm since a while now.

The background is that the MMU emulation on PiStorm (Musashi) is not complete and might break things. 

@ragne1 As you mentioned you give up on that I will close the Issue.

closed 2021-07-30 by captain-amygdala

Markcalibrav6, 2021-07-30:
What you dont understand this is a Work in progress.....Yes somethings dont
work.....however the Developers are working hard to fix issues.....if you
can not wait then the Pistorm is not for you....it simply is not a "plug n
play" device.....so put it in a box and move on for a bit.

captain-amygdala, 2021-07-30:
I think the Issue is closed now. 

So for good, please keep this Issue free from assumptions about @ragne1's perception about the state/compatibility of PiStorm.

I can't reproduce the issue described and for helping we need further information.
I'm happy to reopen the Issue and try to help when ragne1 has further interest in it :)