captain-amygdala/pistorm #47: No problem, but a proposal some easy improvement. (Regarding the 'shut off' problem)

Lalarian, 2021-10-05:
Hello!

I heard from the project on YouTube, from '10 Minute Amiga Retro Cast'.
(https://www.youtube.com/watch?v=u1-QVyw8Tyw)

In the video he said he was in contact with you, and there was an little issue in regards to the RasPi of course not shutting down properly, when the Amiga is reset or shut down.
I had the idea it could make for an useful stability improvement, if your instructions included the installation of '[overlayroot](https://github.com/chesty/overlayroot)', which will effectively make the whole system (with arbitrary exceptions) read-only. It's very easy to setup and would probably avoid potential trouble for most cases, where buffered date would not be written to the storage in time, as the system itself cannot be modified without the user demanding root to be writeable by himself.

paladinprime01, 2021-10-05:
This is a non-issue now. There is a "Shutdown Pi" button in the PiStorm Interaction Tool that allows you to turn off the Pi from Workbench before powering off the computer. No risk of SD card corruption.

Lalarian, 2021-10-05:
Thanks for the info!

Roet-Ivar, 2021-10-05:
> Thanks for the info!

I think this is great idea tbh! If I'm using let's say I use Kickstart 1.3 for nostalgia (or in my case the only rom I legally own atm), I guess powering it down from there is safe for the SD card? Otherwise this seems like a good idea

captain-amygdala, 2021-10-05:
Thanks a lot!

But closing the issue, because 

- So far there was no reported and confirmed case that the SD Card got corrupted by running the Emulation, preventing booting the System afterwards. 
- @paladinprime01 hint about the Shutdown function in the interaction tool
- overlayroot, or a read only rootfs is not a topic on the emulation. But rather on the underlying Raspberry OS 
- not a issue with the emulation itself :)

closed 2021-10-05 by captain-amygdala