captain-amygdala/pistorm #48: Full CDTV support

brufnus, 2021-11-07:
First of all, thanks for this extraordinary project!

I haven't been able to find any info on this subject however, so I'll open a new thread here and hopefully become that wiser (and I know other CDTV owners are keen on this, as well).

Is there currently any work going on with the bus arbitration signals etc, so that CDTV CD-ROM drives and SCSI adapters will become available again? I know the new extended ROM 2.35 supports this, and from what I've been told so far, the issue should be a matter of a firmware update to the CPLD.

This is not meant as a criticism, I know there's probably plenty of stuff to do for you guys, and you've contributed so much to the community already. 
Still, it's always nice to know such things of course, hence the question. c",)

captain-amygdala, 2021-11-08:
Yes, I work on that. It's the so called 'Proto4' CPLD firmware. I can't give a ETA currently, but probably soon. Just to give some insights, the concepts from the Proto4 firmware is also used as foundation for the upcoming PiStorm32 firmware, in the long term this should unify PiStorm and PiStorm32 firmware and low-level raspberry (e.g. the communication protocol to the CPLD/FPGA) development as much as possible.

brufnus, 2021-11-08:
Okay, that's great news! Thanks for a quick reply. c",)