captain-amygdala/pistorm #50: [Makefile:69: emulator] Error 1 on install

peterkomorowski, 2022-01-18:
I get the same error after attempting a "make".  Tried two different SD Cards.  Any suggestions?

Banos1337, 2022-01-26:
> I get the same error after attempting a "make". Tried two different SD Cards. Any suggestions?

Same here. Tried the process from scratch. no luck :(

Banos1337, 2022-01-26:
> I get the same error after attempting a "make". Tried two different SD Cards. Any suggestions?

Read section below:
Important note: If you are using Raspberry Pi OS "Bullseye", the main graphics backend for the OS has changed from dispmanx to DRM, and you need to follow these steps instead of just running make:

captain-amygdala, 2022-01-26:
Some more details would be helpful. Which OS is used, which options on 'make' , unclear parts of the readme/howto so that we can improve it on these parts.

closed 2022-02-09 by captain-amygdala