captain-amygdala/pistorm #57: make problem

wifi75, 2022-06-11:
I don't understand why after a few minutes having given the make command the following error appears:
iq_arm -lasound -lm -lstdc ++
/ usr / bin / ld: cannot find -lbrcmGLESv2
/ usr / bin / ld: cannot find -lbrcmEGL
collect2: error: ld returned 1 exit status
make: *** [Makefile: 69: emulator] Error 1
how come?

LemaruX, 2022-06-11:
Read the instructions and pay attention to the important note section

Important note: If you are using Raspberry Pi OS "Bullseye", the main graphics backend for the OS has changed from dispmanx to DRM, and you need to follow these steps instead of just running make:

First run sudo apt-get install libdrm-dev libegl1-mesa-dev libgles2-mesa-dev libgbm-dev to install the DRM OpenGL/ES libraries, which are for some reason not included with the distro by default. These are necessary to link the graphics output library (raylib).
Then finally, run make PLATFORM=PI3_BULLSEYE for the emulator to compile successfully.

bkodenkt, 2022-06-29:
I had the same problem. This is not mentionend on the walkthrough guides in the Wiki (https://github.com/captain-amygdala/pistorm/wiki/PiStorm-Software-Installation), only on the github main page, see the "simple quickstart" section at https://github.com/captain-amygdala/pistorm

LemaruX, 2022-06-29:
> I had the same problem. This is not mentionend on the walkthrough guides in the Wiki (https://github.com/captain-amygdala/pistorm/wiki/PiStorm-Software-Installation), only on the github main page, see the "simple quickstart" section at https://github.com/captain-amygdala/pistorm

That's because the guide hasn't been updated. The main readme is always up to date.

Kixtart29, 2022-07-17:
Hello,
I'm a beginner with pistorm. I have a problem, same you. 

»
/usr/bin/ld: cannot find -lbrcmGLESv2
/usr/bin/ld: cannot find -lbrcmEGL
collect2: error: ld returned 1 exit status
make: *** [Makefile:69: emulator] Error 1
 »
That the error appear at the end of my configuration...
I write this command:
 »
sudo apt-get install libdrm-dev libegl1-mesa-dev libgles2-mesa-dev libgbm-dev to install the DRM OpenGL/ES libraries

make PLATFORM=PI3_BULLSEYE
 »

It's not running... :/
An idea ?

Bobitsme, 2022-07-17:
Hi, i don't have rpi3 but you can try this :
https://forums.raspberrypi.com/viewtopic.php?t=304614

Kixtart29, 2022-07-18:
[image]

Kixtart29, 2022-07-18:
I think, it's ok for my difficulties.
Now, when I try to start the emulator, I obtain this message: "Failed to open config file default.cfg..."

LemaruX, 2022-07-18:
> I think, it's ok for my difficulties.
> Now, when I try to start the emulator, I obtain this message: "Failed to open config file default.cfg..."
> 
> 

That's because you need to create it. Make a copy the Amiga.cfg file and rename it to default.cfg

The default.cfg file is no longer included, as if it was edited it caused issues when updating (due to the file having been changed). So by creating your own, you can update in future without issues.

Kixtart29, 2022-07-18:
Thank's a lot LemaruX, but :
- How to make a copy of Amiga.cfg and rename it default.cfg ?

I imagine it has to go through a line of code. I do not know how to do.

LemaruX, 2022-07-18:
> Thank's a lot LemaruX, but :
> - How to make a copy of Amiga.cfg and rename it default.cfg ?
> 
> I imagine it has to go through a line of code. I do not know how to do.

The copy command is....

cp amiga.cfg default.cfg

Then make sure to leave amiga.cfg intact, and only make changes in default.cfg

This is also covered in step 15 [here](https://github.com/captain-amygdala/pistorm/wiki/PiStorm-Software-Installation)

Kixtart29, 2022-07-18:
Thank's.
It's look good but the emulator does not start... 

I can't take it anymore... I don't know if I'll get there one day?

Kixtart29, 2022-07-18:
[image]

LemaruX, 2022-07-18:
> Thank's.
> It's look good but the emulator does not start... 
> 
> I can't take it anymore... I don't know if I'll get there one day?

Your screenshot shows the emulator has started.

Have you run buptest to ensure there are not bus communications issues. Details on how to run it can be found in the FAQs on the wiki. Any errors in buptest and it won't work. Often it just needs re-seating in the socket (making sure to push it in all the way, some people seem to not push it hard enough to seat it fully). 

The best way to get support is over on Discord. I suggest you post there if you are still having issues

Kixtart29, 2022-07-18:
That's I think so. But nothing...
I did try to physically remove the pistorm. But nothing helps. By putting the 68k back, it works.
I'm going to start all over again.

LemaruX, 2022-07-18:
> That's I think so. But nothing...
> I did try to physically remove the pistorm. But nothing helps. By putting the 68k back, it works.
> I'm going to start all over again.

And did you run buptest like I said. If that fails, it won't work regardless. Until you run that to ensure there is no bus communication errors, we can't really help you. It's a basic troubleshooting step

Q. What diagnostics can I run to test PiStorm?
A. After flashing the CPLD, compile buptest in the pistorm folder by the following commands:

chmod +x ./build_buptest.sh  
./build_buptest.sh  
sudo ./buptest  
Buptest puts random data into the physical RAM on the amiga so must be run when fully assembled

Kixtart29, 2022-07-18:
Yes, I had run the buptest.

There were no errors.
Then I wanted to try something else and now it doesn't work at all.

I go back to zero.

LemaruX, 2022-07-18:
> Yes, I had run the buptest.
> 
> There were no errors.
> Then I wanted to try something else and now it doesn't work at all.
> 
> I go back to zero.

If buptest actually states 0 errors on the screen then it should work.

Are you getting any output on the native RGB port? 

Try mapping a kickstart ROM

Kixtart29, 2022-07-18:
Too late, I started again.

Nothing in Amiga video output (reason why I put back the original processor with which it works well).

I'll come back to the forum when I'm done.

Kixtart29, 2022-07-18:
Ok

Kixtart29, 2022-07-18:
I'm now on discord.

Kixtart29, 2022-07-18:
[image]

Kixtart29, 2022-07-18:
Ok!
I'm here now.

Kixtart29, 2022-07-18:
Always no RGB output with amiga ???

LemaruX, 2022-07-18:
> Ok!
> I'm here now.

Post over in the #pistorm-classic-support channel on Discord

Kixtart29, 2022-07-18:
I’m always discover DISCORD…

Where and how, I must to post ?
Sorry...

LemaruX, 2022-07-18:
Join the discord server using this invite link. 

https://discord.gg/Gc36ebUXeM

Then post your issue in the support channel.

LemaruX, 2022-08-18:
This can be closed. Not an issue and is covered in the readme.

closed 2022-08-22 by captain-amygdala