captain-amygdala/pistorm #62: Is there a reason why the PiStorm does Not Work in A4000 ?

juanpc2018, 2022-12-19:
very strange Nobody talks about the PiStorm and the A4000 in the same sentence.

captain-amygdala, 2022-12-19:
Yes, the reason is simple :) The current PiStorm 
hardware addresses a 68000 bus interface only. But the A4000 has a 68020/68030 bus interface. 68000 is 16bit, 68020+ is 32bit.  The upcoming PiStorm32 has a compatible bus interface for 32bit Amigas (A1200,CD32,A4000,A3000)

sorlib, 2023-11-29:
Any word on the PiStorm32 for the A4000 and who manages this project?  Thanks Sorli...

captain-amygdala, 2023-11-29:
Yes, the reason is simple :) I have not found time yet . I close the issue again. When I do a A3000/A4000 I will publish it.

Russmrh, 2024-08-01:
how can we better support you in making a a4000 work?

sorlib, 2024-08-01:
Great question, but I'm guessing there isn't a market for A4000 PiStorms?   Maybe one day and who would've thunk the A1200 PiStorm would have ever made landing!  😀

Keep up the great work!  Sorli...

a1exh, 2024-08-01:
I'm sure there is a market. Lots and lots of ReAmiga4000 and AA3000++ to increase the pool size. Yes there are more 060 accelerators for that platform (Legacy + 040->060 adapters + A3660 + BFG9060 + TF4060) PiStorm saves a considerable amount of money expanding an Amiga (RTG + Ethernet + HDD). The A4000 has a nice feature that it has an analog audio mixer on the motherboard intended for CD-ROMs but perfect for PiStorm AHI. The big problem is probably DMA. The original PiStorm and PiStorm32 don't currently have support for multiple DMA masters.

sorlib, 2024-08-01:
Agree completely, my A4000 has so many addons that are now useless (A4000 Coaxial Network, SCSI interface for older A500 Hard Drives, Video Toaster and TBC, and on and on) and that I don't see using.  Back in the day they were very useful, but now they've lost their edge, necessity, and in a box somewhere.  Not familiar with DMA you're referencing, but appreciate the insight and looking forward to future options to bring my A4000 out of storage in the attic.  Personally, I only need a few things including

- Ethernet networking at least for FTP
- USB since no telling where my old Serial Mouse and Keyboard that have come and gone
- and HDMI (still also own my original 1084s monitor but certainly needs recaping and otherwise)
- Acceleration would also be nice from a PiStorm like interface, but when it comes to A4000 native 68040 28mhz is good enough for native software support

Look forward to future options and hopefully my A4000 has died from leaking Capacitors and Battery (pulled long ago)...  Thanks for the reply!  Sorli...

> I'm sure there is a market. Lots and lots of ReAmiga4000 and AA3000++ have been made recently to increase the pool size. Yes there are more 060 accelerators for the A3000/A4000 platform (Legacy + 040->060 adapters + A3660 + BFG9060 + TF4060) PiStorm saves a considerable amount of money expanding an Amiga (RTG + Ethernet + HDD).
> 
> The A4000 has a nice feature that it has an analog audio mixer on the motherboard intended for CD-ROMs but perfect for PiStorm AHI.
> 
> The big problem is probably DMA. The original PiStorm and PiStorm32 don't currently have support for multiple DMA masters.

captain-amygdala, 2024-08-01:
> how can we better support you in making a a4000 work?

Make a day 28 hours long and me 20 years younger ? :)

juanpc2018, 2024-08-01:
> > how can we better support you in making a a4000 work?
> 
> Make a day 28 hours long and me 20 years younger ? :)

very funny...
but the work is already done...
WinUAE has a PowerPC emulator based on QEMU
all code is already done... requires real CyberStorm III Firmware Bios,

latest version of WinUAE does Not work with AmigaOS4, they forked,
only the payd version from ForeverAmiga works,
or older versions with AmigaOS4
HDD controller does Not work.

code is x86_64,
has to be converted/compited to ARM.

captain-amygdala, 2024-08-02:
I'm closing this **issue** again. A PiStorm for A3k/A4k Amigas will happen at one point for sure. For discussion please join our Discord. Spoiler : Most (all?) of us current PiStorm Devs are not interested into PPC Amigas, but @juanpc2018 we are happy to merge a PR to add PPC support. Quoting you 

> > > how can we better support you in making a a4000 work?
> > 
> > 
> > Make a day 28 hours long and me 20 years younger ? :)
> 
> very funny... but the work is already done... WinUAE has a PowerPC emulator based on QEMU all code is already done... requires real CyberStorm III Firmware Bios,
>

closed 2024-08-02 by captain-amygdala