captain-amygdala/pistorm #64: Raspberry Pi 4B problem with make

physcopanda, 2023-01-27:
Hi there - I can't seem to build the software on my Pi for some reason. It's a brand new Pi 4B arrived today. I've done a full upgrade on my pi and installed everything the instructions ask for.

The Raspberry Pi OS I started with was:
Raspberry Pi OS Lite
Release date: September 22nd 2022
System: 64-bit
Kernel version: 5.15
Debian version: 11 (bullseye)

when running make, I get this error:
`
pi@jamie1200pi:~/pistorm $ make
gcc -Wall -Wextra -pedantic -I. -I./raylib -I/opt/vc/include/ -march=armv8-a -mfloat-abi=hard -mfpu=neon-fp-armv8 -O3 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -lstdc++    -c -o emulator.o emulator.c
gcc: error: unrecognized command-line option ‘-mfloat-abi=hard’
gcc: error: unrecognized command-line option ‘-mfpu=neon-fp-armv8’
make: *** [<builtin>: emulator.o] Error 1
`

Hope you can help

LemaruX, 2023-01-27:
Are you using the 64bit version of PiOS? If so you need to switch to using
the 32bit version

physcopanda, 2023-01-27:
ahaa yea thats the problem, many thanks

LemaruX, 2023-03-07:
PiStorm requires 32bit PiOS.

closed 2023-03-07 by LemaruX