captain-amygdala/pistorm #65: Compilation error?

phoyle, 2023-03-04:
Hi, I'm re-setting up a pistorm image after having lost my SD card somewhere and now when I do a 'make' I get the error:

/usr/bin/ld: cannot find -lbrcmGLESv2
/usr/bin/ld: cannot find -lbrcmEGL
collect2: error: ld returned 1 exit status
make: *** [Makefile:69: emulator] Error 1

This is a Pi 3a running Bulleseye 32-bit 2023-02-21 downloaded today. I've copied and pasted the commands from the tutorial on here but no dice.

Any tips?

LemaruX, 2023-03-04:
You haven't read all the instructions in the readme. After the make
command, there is a bit which starts "important note". Read that and follow
the instructions.

phoyle, 2023-03-04:
Ah I wasn't reading that guide but following the one on here which makes no such mention. I fixed it myself anyway with:

sudo ln -s /usr/lib/arm-linux-gnueabihf/libGLESv2.so /usr/lib/libbrcmGLESv2.so
sudo ln -s /usr/lib/arm-linux-gnueabihf/libEGL.so /usr/lib/libbrcmEGL.so

LemaruX, 2023-03-07:
No issue. Instructions in the readme cover this.

closed 2023-03-07 by LemaruX

laubzega, 2023-05-06:
Well, apparently they don't. @phoyle's fix was necessary for me.

LemaruX, 2023-05-06:
> Well, apparently they don't. @phoyle's fix was necessary for me.

Yes they do.

The wiki "install all the things" guide doesn't cover it, but the main readme in the repository does

laubzega, 2023-05-06:
If you mean this bullet:

> First run sudo apt-get install libdrm-dev libegl1-mesa-dev libgles2-mesa-dev libgbm-dev to install the DRM OpenGL/ES libraries, which are for some reason not included with the distro by default. These are necessary to link the graphics output library (raylib).

then no, it is was not sufficient. I got the same error as OP.

LemaruX, 2023-05-06:
I guess they have changed something else in Bullseye since then (though no
one else has reported this anywhere).

Though as Bullseye breaks some other stuff since some of their recent
changes, it's currently advisable to use Buster for the original PiStorm,
and Bullseye for the PiStorm32-Lite.

drphilxr, 2024-01-14:
this ^ exact sequence happened to me as well following the main git directions...thankyou for listing the fixes

kawalkQCWski, 2025-03-23:
Instructions not the best ... :) but glad to have these