captain-amygdala/pistorm #66: --config-file option does not seem to be implemented in pistorm32 branch

TonyHoyle, 2023-03-18:
--config-file option does not work.  

in boot_scripts/README.md the documentation states:

If you wish to boot using a custom configuration file change `ExecStart` to:

```ini
ExecStart=/home/pi/pistorm/emulator --config-file myconfig.cfg
```

This does not seem to be implemented currently in the pistorm32 branch.. instead it tries to open a file called '--config-file' instead and crashes.

root@pistorm:/opt/pistorm# /opt/pistorm/emulator --config-file config/pistorm.cfg
FPGA Firmware : --config-file
Segmentation fault

Or, via strace:

fstat64(1, {st_mode=S_IFCHR|0600, st_rdev=makedev(0x88, 0), ...}) = 0
write(1, "FPGA Firmware : --config-file\n", 30FPGA Firmware : --config-file
) = 30
openat(AT_FDCWD, "--config-file", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault

It's possible the documentation is out of date and there is a different command?

LemaruX, 2023-03-18:
As a workaround, if you call your config default.cfg then you don't need to
specify it at all.

Guddler, 2023-04-06:
@TonyHoyle I don't know what the official answer is but I suspect someone modified emulator and didn't update the documentation? It seems to want the name of the firmware as the 1st parameter when supplying options. Since the default firmware appears to be 'bitstream.bin', the following worked for me

`./emulator bitstream.bin --config ../amiga.cfg`