captain-amygdala/pistorm #68: Update documentation for OpenOCD

phalkon13, 2023-05-08:
The newest version of OpenOCD seems to be not working properly with PiStorm, you get the following error:
"Could not detect CPLD"

Per the Discord, they have a fix:
https://discord.com/channels/784428461330530305/833686489674154014/1092141565918593206

Installing version 0.11.0~rc2-1 seems to work:
sudo apt-get install openocd=0.11.0~rc2-1

I would recommend updating your documentation to reflect that for ease of use.

(Aside from that, your documentation is pretty wonderfully straight-forward, thank you very much for that!)

venice1200, 2023-09-03:
I can confirm the fix running raspios bullseye 2023-05-03.

closed 2023-10-12 by captain-amygdala

davidbhowmik, 2023-10-30:
I am trying to install Pistorm on a Raspberry Pi 3B+. I am using Rapian Bullseye 32 bit Lite. I have tried: sudo apt-get install openocd=0.11.0~rc2-1. And have tried both ./flash.sh and ./flash_experimental.sh but still get "Could not detect CPLD"

any help please?

LemaruX, 2024-01-19:
> Same problem.
> 
> I confirm, new install (2023-12-11-raspios-bookworm-armhf-lite.) and follow the github for pistorm flashing **Don't work.** **Could not detect CPLD**

You can also flash the firmware from this repo here. It uses a new flashing tool so doesn't suffer this issue

[https://github.com/LemaruX/PiStorm-Firmware](url)

Gnome78, 2024-01-19:
Ths a lot, it's woking.