captain-amygdala/pistorm #71: Kickstart 34.005 (1.3) RAD Drive Reboot

nyteshade, 2023-08-26:
I have a PiStorm classic running in an A1000. When I set the config for musashi to use kickstart 34.005 or 1.3, and then boot to workbench, if I create a RAD drive reboots are usually not survived. Neither is softkicking into 1.3 from 3.2.2.1. It literally has no effect. Is there a setting I can use to fix this? Are booting from autoconfig hdds in 1.3 supported yet?

LIV2, 2023-08-26:
Kickstart 1.3 has a bug that prevents RAD drives from working in machines with more than 512K memory.
To fix this you will want to add 
`setpatch r`
To your startup-sequence 

1.3 does support booting from HDDs however afaik piscsi itself does not support 1.3

nyteshade, 2023-08-27:
Would love to know if piscsi ever attains 1.3 support. It will change a lot of things. I'll give the `setpatch r` a chance, though I quite specifically remember using RAD disks on my A500 with more than 1MB of RAM.

vjouppi, 2023-09-10:
The setpatch r workaround is only needed if you have 1MB of Chip RAM or more under 1.3.

The amount of Fast RAM does not affect reset surviving resident modules.

closed 2023-10-12 by captain-amygdala