captain-amygdala/pistorm #72: PiOS Lite 32 Bookworm

DoubleA-WTH, 2023-10-12:
Hi,

it seems, that the a314 part doesn't compile with the current piOS lite Bookworm edition.
Maybe some more packets are missing in the default setup for bookworm?

Thanks for your time and this project
Andreas

captain-amygdala, 2023-10-12:
Hi! Thanks for reporting. I will look into it in a couple of days

Tsthanos, 2023-10-31:
Hello I have the same problem with a raspberry pi zero 2w and bookworm version

LemaruX, 2023-10-31:
> Hello I have the same problem with a raspberry pi zero 2w and bookworm version

Bookworm is not yet supported. Stick to using Bullseye

Chandler-Kluser, 2023-12-30:
in my case I compiled with raspi OS 32-bit Bullseye and it did not compile with `make`

I was able to build it with `make PLATFORM=RPI3_BULLSEYE`, but it didn't work with my A500 (purple screen, no boot...)

Chandler-Kluser, 2024-05-13:
I was able to build running only `make` in Raspberry Pi OS Lite (32-bits) in my Raspberry Pi Zero 2W making these modifications:

```
sudo ln -s /usr/lib/arm-linux-gnueabihf/libGLESv2.so /usr/lib/arm-linux-gnueabihf/libbrcmGLESv2.so
sudo ln -s /usr/lib/arm-linux-gnueabihf/libEGL.so /usr/lib/arm-linux-gnueabihf/libbrcmEGL.so
```

to overcome this link error:
```
gcc -o emulator emulator.o memory_mapped.o config_file/config_file.o config_file/rominfo.o input/input.o gpio/ps_protocol.o platforms/platforms.o platforms/amiga/amiga-autoconf.o platforms/amiga/amiga-platform.o platforms/amiga/amiga-registers.o platforms/amiga/amiga-interrupts.o platforms/mac68k/mac68k-platform.o platforms/dummy/dummy-platform.o platforms/dummy/dummy-registers.o platforms/amiga/Gayle.o platforms/amiga/hunk-reloc.o platforms/amiga/cdtv-dmac.o platforms/amiga/rtg/rtg.o platforms/amiga/rtg/rtg-output-raylib.o platforms/amiga/rtg/rtg-gfx.o platforms/amiga/piscsi/piscsi.o platforms/amiga/ahi/pi_ahi.o platforms/amiga/pistorm-dev/pistorm-dev.o platforms/amiga/net/pi-net.o platforms/shared/rtc.o platforms/shared/common.o m68kcpu.o m68kdasm.o softfloat/softfloat.o softfloat/softfloat_fpsp.o m68kops.o a314/a314.o -O3 -pthread -Wall -Wextra -pedantic -L/opt/vc/lib -L./raylib -lraylib -lbrcmGLESv2 -lbrcmEGL -lbcm_host -lstdc++ -lvcos -lvchiq_arm -lasound -lm -lstdc++
/usr/bin/ld: cannot find -lbrcmGLESv2
/usr/bin/ld: cannot find -lbrcmEGL
collect2: error: ld returned 1 exit status
make: *** [Makefile:69: emulator] Error 1
```

closed 2024-05-15 by captain-amygdala