captain-amygdala/pistorm #78: Pistorm Emulator crashes when I install Roadshow for internet

Tsthanos, 2024-02-28:
I followed the instructions for networking my amiga 2000 with WB 3.2 with Roadshow but after rebooting emulator crashes constantly.  This is the  error message in the raspberry CLI.
[image]

Victory2018, 2024-03-01:
I sorted this on mine, it was something to do with the OS changes on the Pi with networking, i fixed it by adding the following two lines in etc/rc.local just before the line exit 0. Then reboot the pi.

sudo ip tuntap add mode tap tap0
sudo ifconfig tap0 192.168.2.1 up

closed 2024-03-03 by Tsthanos

theonetruetom, 2024-06-03:
This works, if you're not getting failures in rc-local compatibility from systemd.  I had to do something a bit more convoluted and add those two commands as a script in ~ called tuntap.sh: 
```
#!/bin/bash

ip tuntap add mode tap tap0
ifconfig tap0 192.168.2.1 up

```
and then /etc/systemd/system/pi-tuntap.service:

```
[Unit]
Description=PiStorm tuntap device setup
Wants=network-online.target
After=network-online.target network.target

[Service]
ExecStart=/home/<username>/tuntap.sh

[Install]
WantedBy=multi-user.target

```