captain-amygdala/pistorm #79: FEATURE REQUEST : Microchip ENC28J60 Ethernet module support

dlasher, 2024-06-09:
While I realize there's already "host" ethernet support (from the A314 project) given how the pistorm "maps" devices through, this would be a fantastic candidate for inclusion. Now the amiga can get it's own dedicated ethernet interface independent of the PI.

### Microchip ENC28J60 Ethernet module
* CHEAP - ready to go boards are $5 in quantity on amazon
* can be driven off the GPIO ports on the PI
* already has an ethernet driver compiled by the Apollo/Vampire guys (v2expeth.device)
* source code for driver freely available : https://gitlab.com/MHeinrichs/Zorro-LAN-IDE/-/tree/master/Driver/baxnet/source/sdnet

### REFERENCES
* https://www.instructables.com/Super-Cheap-Ethernet-for-the-Raspberry-Pi/
* https://majsta.com/modules.php?name=News&file=article&sid=179
* https://gitlab.com/MHeinrichs/Zorro-LAN-IDE/tree/master/Driver/baxnet
* https://gitlab.com/MHeinrichs/Zorro-LAN-IDE/-/tree/master/Driver/baxnet/source/sdnet

Thanks for consideration.

captain-amygdala, 2024-06-09:
Unfortunately, there are no free GPIO pins on the Raspberry with Pistorm. All pins are needed for the 68k hardware bus emulation.

closed 2024-06-09 by captain-amygdala

captain-amygdala, 2024-06-12:
Yes sure, any ethernet/WiFi/5g/usb rndis/whatever usb adapter with Linux drivers can be used as long as it's a normal Linux network interface