captain-amygdala/pistorm #81: Build fails due to missing header files

SvenMichaelKlose, 2024-09-28:
Latest Raspi OS Lite 32-bit on RPI-3A:

gcc -Wall -Wextra -pedantic -I. -I./raylib -I/opt/vc/include/ -march=armv8-a -mfloat-abi=hard -mfpu=neon-fp-armv8 -O3 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -lstdc++    -c -o platforms/amiga/rtg/rtg-output-raylib.o platforms/amiga/rtg/rtg-output-raylib.c
platforms/amiga/rtg/rtg-output-raylib.c:17:10: fatal error: interface/vmcs_host/vc_tvservice.h: No such file or directory
   17 | #include "interface/vmcs_host/vc_tvservice.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: platforms/amiga/rtg/rtg-output-raylib.o] Error 1

Skelbagz, 2024-10-05:
There is a comment in the README.md file which addresses this issue. I would also make sure (if you are using the RPi imager) to select **Raspberry Pi OS (Legacy, 32 bit) Lite** as the ISO to use.

> **Important note:** If you are using **Raspberry Pi OS "Bullseye"**, the main graphics backend for the OS has changed from dispmanx to DRM, and you need to follow these steps instead of just running `make`:
* First run `sudo apt-get install libdrm-dev libegl1-mesa-dev libgles2-mesa-dev libgbm-dev` to install the DRM OpenGL/ES libraries, which are for some reason not included with the distro by default. These are necessary to link the graphics output library (raylib).
* Then finally, run `make PLATFORM=PI3_BULLSEYE` for the emulator to compile successfully.

SvenMichaelKlose, 2024-10-06:
Thanks! I used that image. It's a friend's dream to have that thing installed, so I'll have to check it out later.

swindels, 2024-10-16:
I had the same issue using the "Latest Raspi OS Lite 32-bit on RPI-3A". This solution didn't work as the latest version is Bookworm. My resolution was to start again using Bullseye Lite (Now Legacy) and then build work as described

StrayDogge, 2025-01-02:
The latest Raspberry PI OS image - as of writing this - is 12 (bookworm). It has the `/opt/vc/` folder removed and hence any user will run into this same issue. I had to reinstall the OS image using the legacy bullseye version to successfully compile the emulator.

JoshS80, 2025-01-14:
For anyone using Raspberry Pi OS 12 (bookworm), the failure finding header file `vc_tvservice.h` can be resolved by adding `libraspberrypi-dev` to the apt-get command line used for Raspberry Pi OS 11 (bullseye). So the apt-get command line would become `sudo apt-get install libdrm-dev libegl1-mesa-dev libgles2-mesa-dev libgbm-dev libraspberrypi-dev`. Then run the make command as already indicated for "bullseye".

Doing this I was able to get the build working with "bookworm" and running the emulator binary seems to work properly. If needed I could submit a PR to update the readme.

codewiz, 2025-07-04:
This build error is still present when using the latest 32-bit Raspberry Pi OS image. The change proposed in the previous comment works for me.

Could someone please send a PR to update README.md accordingly? Or better, update the Makefile to work with Bookworm by default.

JorgeHawkins, 2025-10-29:
The following error raises when trying `sudo apt-get install libraspberrypi-dev` with default RPi Imager settings:

`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libraspberrypi-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libdtovl0:arm64 libdtovl0

E: Package 'libraspberrypi-dev' has no installation candidate`

ovesennet, 2025-11-06:
> You won't have a great experience trying to use floppies with a PiStorm (or on many other accelerators for that matter). You will find that many will run too slow/fast or now at all.
> 
> You really want to be running them from the SD card (or worse case HDD or CF Card) using WHDLoad. We have a tool called the WHDLoad Wrapper which makes games run properly on a pistormed machine by applying the relevant emulator and WHDLoad settings.
> 
> Instead of using the Imager tool, you can also just write the Emu68 files to a FAT32 formatted SD card, though ideally you still want a kick ROM on there (A1200 version regardless of what Amiga model you have). That will make the PiStorm just as an accelerator and ram expansion and nothing else. But running from floppy still won't be great. You can get the files here https://github.com/michalsc/Emu68
> 
> You may be better off just writing CaffeineOS to an SD card. It's a ready made image that contains a complete game set and other tools and programs. You just write it to an SD card and you are good to go.
> 
> Unfortunately we cannot share the link to CaffeineOS here, as it contains an OS, ROMs and games/programs (all of which are copyright). It shouldn't be hard to find though.
> 
> Or, if you really want to run from floppy, you would probably be better off with a stock machine with just a ram expansion to be honest.

Thanks for the advice! Yes, I will look into that. For now I just wanted to try to install it as basic as possible to see what happens to my bare bones Amiga 500 and to check out the speed bench mark in Sysinfo etc.

mja65, 2025-11-06:
If you use the Emu68 Imager you can still boot from a floppy disk if you really want to . As for the kickstart Rom and Workbench files - you need the Kickstart ROM in order to boot PiStorm anyway.

gianlucarenzi, 2025-11-07:
The main issue here using emu68 imager there is no Linux alternative. It
relais on the Windows OS.
To my knowledge supporting musashi it is a great option to me, having Linux
under the hood even if it has less powerhorses...

Ciao e buona giornata.

"GP! In mezzo al campo stai proprio schifoso!"
Coach M.Russo

LemaruX, 2025-11-07:
You don't need to use the imager to use emu68, you can manually install it
the same as everyone did before the imager was created.

There is no need to use Musashi just because you don't run Windows.

nygellyndley, 2026-08-15:
> Emu68 is much more compatible, and is actively being developed.

Not so compatible if you don't run windows though.

LemaruX, 2026-08-15:
> > Emu68 is much more compatible, and is actively being developed.
> 
> Not so compatible if you don't run windows though.
> 
> 

You don't need windows to use Emu68, so not sure what you are referring to.

Just prep the SD card and copy the emu68 files onto it

https://michalsc.github.io/Emu68/tutorials/